matthias-bs/BresserWeatherSensorLW #4: Request for testing: Heltec WiFi LoRa 32 V3

matthias-bs, 2024-04-16:
@evon800c You kindly offered your help in testing - much appreciated!

It would be great if you could test if the current main branch works. Heltec WiFi LoRa 32 V3 is implemented, even if the README.md doesn't show it.

evon800c, 2024-04-16:
HI @matthias-bs,

I have downloaded your code, extract files, change LoRaWan date (for my personal LoRaWan network with ChirpStack), change sleep time from 900s to 120s, set verbose mode and compile.
I have used arduino ide, (if you want i can use vscode).
in attach the serial output. and bin

Davide
[20240416.01.zip](https://github.com/matthias-bs/BresserWeatherSensorLW/files/14999752/20240416.01.zip)

matthias-bs, 2024-04-16:
@evon800c 
Hi Davide,

the ESP goes into sleep mode right away, because it seems the battery voltage is low.

From the schematic:
* GPIO1 is the ADC input (ADC_IN)
* The input divider is R14 (390K) to measured voltage and R17 (100K) to GND (ADC_IN in between R14 & R17)
* GPIO37 (ADC_Ctrl) and GPIO36 (Vext_Ctrl) somehow control if and which voltage is measured

Maybe you can find a source code example on how to get the voltage?

I have changed src/adc/adc.cpp to return 0 (not able to measure) to get it out of the way for the moment. I already set the input pin in BresserWeatherSensorLWCfg.h, but the divider still has to be set too.

I'm too tired to complete this right now, but with the change above you should be able to get one step further.

Best regards,
Matthias

evon800c, 2024-04-16:
Hi @matthias-bs,

I tried with the change you inserted and if I understand the code correctly, it is on line 82, where if a "`#elif defined(ARDUINO_heltec_wifi_32_lora_V3)`" board is detected the return value is set to 0 to exclude the charge evaluation of the battery.
Unfortunately the board is generally recognized as "`ARDUINO_heltec_wifi_lora_32_V3`" and not as "`ARDUINO_heltec_wifi_32_lora_V3`".

I then tried to change line 82 as "`#elif defined(ARDUINO_heltec_wifi_32_lora_V3) || defined(ARDUINO_heltec_wifi_lora_32_V3)`", but nothing has changed.

I tried to insert some messages for debugging and I saw that in reality the "`ADC_EN`" path was chosen. I tried putting a return 0 there, but nothing changed.
Then going backwards, I modified the code of the previous function (`getVoltage`) so that it returned a sufficient value (fictitiously) to continue, but it goes in panic and restart.

I suppose that heppen because it can't identify the correct board and pins for the radio module.
```
00:30:16.700 -> [  2680][I][BresserWeatherSensorLW.ino:435] setup(): 
00:30:16.700 -> Setup
00:30:16.700 -> [  2686][I][BresserWeatherSensorLW.ino:211] print_wakeup_reason(): Wake not caused by deep sleep: 0
00:30:16.727 -> [  2695][I][BresserWeatherSensorLW.ino:463] setup(): Boot count: 1
00:30:16.727 -> [  2701][I][BresserWeatherSensorLW.ino:301] printDateTime(): 1970-01-01 01:14:12
00:30:16.727 -> [  2708][V][Preferences.cpp:376] getUShort(): nvs_get_u16 fail: sleep_int NOT_FOUND
00:30:16.727 -> [  2716][D][BresserWeatherSensorLW.ino:471] setup(): Preferences: sleep_interval:        360 s
00:30:16.727 -> [  2724][V][Preferences.cpp:376] getUShort(): nvs_get_u16 fail: sleep_int_long NOT_FOUND
00:30:16.759 -> [  2732][D][BresserWeatherSensorLW.ino:473] setup(): Preferences: sleep_interval_long:   60 s
00:30:16.759 -> [  2741][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type ADC_ONESHOT (7) successfully set to 0x420283ec
00:30:16.759 -> [  2752][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 4 successfully set to type ADC_ONESHOT (7) with bus 0x5
00:30:16.791 -> [  2763][D][esp32-hal-adc.c:309] __analogReadMilliVolts(): Creating cali handle for ADC_0
00:30:16.791 -> [  2771][D][adc.cpp:68] getVoltage(): Voltage = 3610mV
00:30:16.791 -> [  2741][D][adc.cpp:102] getBatteryVoltage(): ddasaro: ADC_EN: {3610}
00:30:16.791 -> [  2783][D][adc.cpp:68] getVoltage(): Voltage = 3320mV
00:30:16.791 -> [  2788][D][adc.cpp:68] getVoltage(): Voltage = 2890mV
00:30:16.832 -> [  2793][D][adc.cpp:68] getVoltage(): Voltage = 2850mV
00:30:16.832 -> [  2793][D][adc.cpp:102] getBatteryVoltage(): ddasaro: ADC_EN: {2850}
00:30:16.832 -> [  2805][D][adc.cpp:68] getVoltage(): Voltage = 3340mV
00:30:16.873 -> I NimBLEDevice: BLE Host Task Started
00:30:16.873 -> I NimBLEDevice: NimBle host synced.
00:30:16.873 -> D NimBLEScan: >> start: duration=31
00:30:16.873 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
```

I'm also looking for a code that i have used some time ago to check battery level
if you don't have any suggestion, tomorrow i try to understand why the board is not correctly evaluate.

thanks Davide.

evon800c, 2024-04-17:
Hi @matthias-bs ,

this morning I did some checks and I realized that I made the same mistake as last time. I used the wrong library boards.
I did a quick check and test without LoRaWan GW.
Now I see the situation completely differently.
I hope this evening to be able to do some more in-depth tests to observe the dialogue to and from LoRaWan and in receiving data from the weather station.

I hope to be able to update you in the evening.

Davide

evon800c, 2024-04-17:
Hi @matthias-bs,

i'm at home.
I have downloaded your code, put info for my lorawan and burn... on first check all it seems ok.
in attach a file with serial monitor output on degub level (if you need verbose level, let me know) for first 2 boot cycles. now i need to check which is the best decode in your "scripts" folder that can me ok for me....
i have a simple private lorawan in my personal lab, implemented with chirpstack that is as like as ttn.

let me know if you need some particular check from my side.
in the meanwhile i'll try to looking code regarding battery level check.

Davide

[20240417.01.txt](https://github.com/matthias-bs/BresserWeatherSensorLW/files/15016555/20240417.01.txt)

matthias-bs, 2024-04-17:
Hi @evon800c ,

That's good news! Yes, everything fine in the log file! Thank you!

I have to update the scripts anyway, but only the command downlink/response uplink parts will be affected.
A new release is on its way...

Regards,
Matthias

matthias-bs, 2024-04-18:
BTW: Please expect the command/status messages to change in the near future. I think I will use the FPort in commands instead of the first payload byte for multiplexing.

evon800c, 2024-04-18:
OK,
in the next few days i will try to analyze and better understand your code.
i found something for measuring the battery, but before writing something to you I want to find out more.
if you need other some specific test, feel free to contact me.

matthias-bs, 2024-04-19:
Hi @evon800c ,

most of the time I'm using Visual Studio Code with the Arduino extension - I don't want to miss it any more.

In certain cases I still have to use the Arduino IDE:
- Show detailed compiler output -  I still have to find out how to do this with vscode
- Serial monitor - the one integrated in vscode is broken since a few days (at least for me) :unamused:
- Exception decoder - hopefully rarely needed

Cheers
Matthias

evon800c, 2024-04-19:
@matthias-bs,

if i well understand you use "Visual Studio Code extension for Arduino" and not "PlatformIO IDE for VSCode".
I never use Arduino IDE on vscode, i'll try.
With PlatformIO, the heltec wifi lora 32 v3 board is view as:
`**ARDUINO_heltec_wifi_lora_32_V3**` and not as 
`**ARDUINO_heltec_wifi_32_lora_V3**` as in your code.

in platformio.ini file, the code is:
```
[env:heltec_wifi_lora_32_V3]
platform = espressif32
board = heltec_wifi_lora_32_V3
framework = arduino
```
I don't know if i'm doing some error....

This info is only to share with your what i see. No problem form me, i only need to add an "OR" in boards evaluation.
I don't know if this info can be helpfull for you and others.

Thans Davide.

matthias-bs, 2024-04-19:
@evon800c 
no, I don't use PlatformIO. It's sad but true, there several different versions of the board defines, even within the Arduino board ESP32 package. The maintainers do their best, but still... E.g. things have been modified while the board package version still remained the same. The fact that board manufacturers tend to create their own packages (not always the shape you could hope for) instead of contributing to Arduino-ESP32 does not help to improve the situation, either.

If you have a variant different from the one I am using, please let me know.

evon800c, 2024-04-20:
Hi @matthias-bs ,

I did some tests with "Arduino iDE" and "vsCode" using "PlatformIO IDE".

With "Arduino IDE", using the "`esp32`" library and the board: "`Heltec WiFi Lora 32(V3) / Wireless shell(V3) / Wireless stick lite (V34)`", your code, both in "`BresserWeatherSensorLW`" and in "`BresserWeatherSensorReceiver`" library, correctly recognizes the board.
With "vsCode" however, using the "`espressif32`" library and selecting the "`Heltec WiFi LoRa 32 (V3)`" board, the board is not recognized.

I use the ["Heltec WiFi LoRa 32(V3)](https://heltec.org/project/wifi-lora-32-v3/)" board which is different from "[Heltec Wireless Stick(V3)](https://heltec.org/project/wireless-stick-v3/)".

I solve the problem by inserting in your code next to "`defined(ARDUINO_heltec_wifi_32_lora_V3)`" an `or` with "`defined(ARDUINO_heltec_wifi_lora_32_V3)"`. This way everything works correctly.

If this is not a problem for you, in your source code, insert both names, so at least for these two development IDEs the board will always be identified.

I change the code in:
- `WeatherSensorCfg.h` of the library i made change in line 213 and 477
- `BresserWeatherSensorLWCfg.h` i made change in line 316
- `adc.cpp` i made change in line 82

Davide

matthias-bs, 2024-04-20:
Hi @evon800c,

thanks for testing! Yes, I will add this to the code. Or if you have your modified code in a Github branch, you can just make a pull request.

Cheers,
Matthias

evon800c, 2024-04-20:
@matthias-bs,
I have no made branch, only a simple change on my local code.
Davide

matthias-bs, 2024-04-20:
@evon800c Just updated the uplink for matter on branch https://github.com/matthias-bs/BresserWeatherSensorLW/tree/fix-payload-formatters - beware: not tested much...

evon800c, 2024-04-22:
hi @matthias-bs,

I don't know if it can be useful to you, but I found this link interesting.
I think that as soon as I find a moment I will try to go deeper and do some tests.

http://digitalconcepts.net.au/arduino/index.php?op=Battery#wifilora3

evon800c, 2024-04-24:
Hi, @matthias-bs

I saw that you also released a new version of the "`BresserWeatherSensorReceiver`" library.
Were you able to insert the evaluation of both identities of the "Heltec WiFi LoRa 32 V3" here too?

- "`defined(ARDUINO_heltec_wifi_32_lora_V3)`"
- "`defined(ARDUINO_heltec_wifi_lora_32_V3)`"

Davide

matthias-bs, 2024-04-27:
Hi @evon800c ,

I just pushed a new release v0.27.2 of BresserWeatherSensorReceiver with the updated defines.
Could you please test the battery voltage measurement from the feature branch https://github.com/matthias-bs/BresserWeatherSensorLW/tree/feature-heltec-v3-battery-voltage?

Best regards,
Matthias

evon800c, 2024-04-30:
Hi, @matthias-bs 
first test is not gone in the right way.
i have only get code and update.
i have also set the new BresserWeatherSensorReceiver version.

```
[env:heltec_wifi_lora_32_V3]
platform = espressif32
board = heltec_wifi_lora_32_V3
framework = arduino
lib_deps = 
	matthias-bs/BresserWeatherSensorReceiver@^0.27.2
	jgromes/RadioLib@^6.5.0
	thesolarnomad/LoRa Serialization@^3.2.1
	fbiego/ESP32Time@^2.0.6
	h2zero/NimBLE-Arduino@^1.4.1
	https://github.com/theengs/decoder
build_flags = 
	;-DCORE_DEBUG_LEVEL=0		; None
	;-DCORE_DEBUG_LEVEL=1		; Error
	;-DCORE_DEBUG_LEVEL=2		; Warn
	;-DCORE_DEBUG_LEVEL=3		; Info
	;-DCORE_DEBUG_LEVEL=4		; Debug
	-DCORE_DEBUG_LEVEL=5		; Verbose
```

panic

```
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4206d442
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[  2211][I][BresserWeatherSensorLW.ino:473] setup(): Setup
[  2212][I][BresserWeatherSensorLW.ino:232] print_wakeup_reason(): Wake not caused by deep sleep: 0
[  2214][I][BresserWeatherSensorLW.ino:501] setup(): Boot count: 1
[  2221][I][BresserWeatherSensorLW.ino:326] printDateTime(): 1970-01-01 02:15:20
[  2228][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: ble NOT_FOUND
[  2235][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: ble NOT_FOUND
[  2243][D][AppLayer.h:191] bleAddrInit(): Using BLE addresses from BresserWeatherSensorLWCfg.h:
[  2251][D][AppLayer.h:200] bleAddrInit(): a4:c1:38:b8:1f:7f
[  2257][V][Preferences.cpp:376] getUShort(): nvs_get_u16 fail: sleep_int NOT_FOUND
[  2264][D][BresserWeatherSensorLW.ino:516] setup(): Preferences: sleep_interval:        360 s
[  2272][V][Preferences.cpp:376] getUShort(): nvs_get_u16 fail: sleep_int_long NOT_FOUND
[  2280][D][BresserWeatherSensorLW.ino:518] setup(): Preferences: sleep_interval_long:   900 s
[  2389][I][esp32-hal-adc.c:243] __analogReadMilliVolts(): ADC1: Characterized using Default Vref: 0

[  2390][D][adc.cpp:74] getVoltage(): Voltage = 7804mV
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42007b15  PS      : 0x00060830  A0      : 0x82007b10  A1      : 0x3fcebf70  
A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000061  
A6      : 0x3c07069f  A7      : 0x00000001  A8      : 0x82028ea1  A9      : 0x3fcebf30  
A10     : 0x00000000  A11     : 0x00000031  A12     : 0x00000031  A13     : 0x00000000  
A14     : 0x00000020  A15     : 0x00001e7c  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000005  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  

Backtrace: 0x42007b12:0x3fcebf70 0x42007b0d:0x3fcebf90 0x42003ed0:0x3fcebfb0 0x4202a11e:0x3fcec080
```

i'll promise you that i'll do, some deeper test.

davide

evon800c, 2024-04-30:
Hi, @matthias-bs

I think that i found the panic cause. 
In adc.cpp file, line 90 need to be changed from ```getVoltage();``` to ```return getVoltage();```

Davide

matthias-bs, 2024-05-01:
> Hi, @matthias-bs
> 
> I think that i found the panic cause. In adc.cpp file, line 90 need to be changed from `getVoltage();` to `return getVoltage();`
> 
> Davide

Arrgh! You're right! I fixed that.

matthias-bs, 2024-05-02:
@evon800c Thank you very much for testing and for providing the link to the very helpful article about the circuit implementation!

closed 2024-05-02 by matthias-bs