matthias-bs/BresserWeatherSensorLW #94: failing to join lorawan and other issues!

mczakk, 2024-08-09:
Hi Matthias, 
I have been trying to connect my esp32 using this repository as i saw that BresserWeatherSensorTTN was being deprecated.
I can complie the code, but it fails to join the lorawan network. Your code calls for a nwk key, but on TTN i cvan only see nwkskey.
I also have other questions...

mczakk, 2024-08-12:
OK, i've resolved one issue, the device had to be re registered as a lorawan version 1.1, this now shows me a nwk key, but still wont join 
`11:56:43.488 -> [183476][V][BresserWeatherSensorLW_Aug_24.ino:593] setup(): Initalise the radio
11:56:43.522 -> [183502][D][BresserWeatherSensorLW_Aug_24.ino:393] lwActivate(): Recalling LoRaWAN nonces & session
11:56:43.522 -> [183513][W][config.h:402] debug(): Restoring nonces buffer failed - RADIOLIB_LORAWAN_NONCES_DISCARDED (-1119)
11:56:43.522 -> [183524][W][config.h:402] debug(): Restoring session buffer failed - RADIOLIB_ERR_NETWORK_NOT_JOINED (-1101)
11:56:43.522 -> [183533][D][BresserWeatherSensorLW_Aug_24.ino:433] lwActivate(): Join ('login') to the LoRaWAN Network
11:56:50.642 -> [190646][D][BresserWeatherSensorLW_Aug_24.ino:437] lwActivate(): Saving nonces to flash
11:56:50.754 -> [190655][D][BresserWeatherSensorLW_Aug_24.ino:447] lwActivate(): Join failed: -1116`

is there a setting i'm missing?

matthias-bs, 2024-08-12:
Did you look up the error code in RadioLib?

mczakk, 2024-08-13:
the only error code listed is -1116 'no downlink after uplink', heres the serial monitor lines:
`09:36:42.375 -> [183471][V][BresserWeatherSensorLW_Aug_24.ino:593] setup(): Initalise the radio
09:36:42.407 -> [183497][D][BresserWeatherSensorLW_Aug_24.ino:393] lwActivate(): Recalling LoRaWAN nonces & session
09:36:42.407 -> [183508][W][config.h:402] debug(): Restoring nonces buffer failed - RADIOLIB_LORAWAN_NONCES_DISCARDED (-1119)
09:36:42.439 -> [183519][W][config.h:402] debug(): Restoring session buffer failed - RADIOLIB_ERR_NETWORK_NOT_JOINED (-1101)
09:36:42.439 -> [183528][D][BresserWeatherSensorLW_Aug_24.ino:433] lwActivate(): Join ('login') to the LoRaWAN Network
09:36:49.563 -> [190641][D][BresserWeatherSensorLW_Aug_24.ino:437] lwActivate(): Saving nonces to flash
09:36:49.563 -> [190650][D][BresserWeatherSensorLW_Aug_24.ino:447] lwActivate(): Join failed: -1116
09:36:49.563 -> [190657][I][BresserWeatherSensorLW_Aug_24.ino:454] lwActivate(): Boots since unsuccessful join: 354
09:36:49.563 -> [190666][I][BresserWeatherSensorLW_Aug_24.ino:455] lwActivate(): Retrying join in 180 seconds`

more fundamentally the code now won't compile, give the error :
`/home/matt/Arduino/Sketches/BresserWeatherSensorLW_Ll1/BresserWeatherSensorLW_Aug_24/BresserWeatherSensorLW_Aug_24.ino: In function 'int16_t lwActivate()':
/home/matt/Arduino/Sketches/BresserWeatherSensorLW_Ll1/BresserWeatherSensorLW_Aug_24/BresserWeatherSensorLW_Aug_24.ino:391:3: error: 'node' was not declared in this scope
  391 |   node.beginOTAA(joinEUI, devEUI, nwkKey, appKey);
      |   ^~~~`
i tried compiling the bare code with no modifications and it doesn't like that either, won't find 'logging.h', even though it's in the correct place!

mczakk, 2024-08-13:
OK, so i discovered that i was missing a config.h file and the 'node not declared' error is now gone. 
Now i get the error:
```/home/matt/Arduino/Sketches/BresserWeatherSensorLW_Aug_24/secrets.h:9:42: error: expected unqualified-id before numeric constant
    9 | #define RADIOLIB_LORAWAN_DEV_EUI   0x70, 0xB3, 0xD5, 0x7E, 0xD0, 0x06, 0x9B, 0x1C```
chatgpt suggests that his is because the array is declared incorrectly, and that i should define the values in an array in the 'secrets.h' file, buty tyrying to do this just causes loads more problems

mczakk, 2024-08-13:
Next update:-
had to amend the config.h file 
```
// Copy over the EUI's & keys in to the something that will not compile if incorrectly formatted
uint64_t joinEUI[8] =   {RADIOLIB_LORAWAN_JOIN_EUI};
uint64_t devEUI[8]  =   {RADIOLIB_LORAWAN_DEV_EUI};
uint8_t appKey[16] = { RADIOLIB_LORAWAN_APP_KEY };
uint8_t nwkKey[16] = { RADIOLIB_LORAWAN_NWK_KEY };
```
and the call to node.beginOTAA:
```
 // setup the OTAA session information
  node.beginOTAA((uint64_t)joinEUI, (uint64_t)devEUI, nwkKey, appKey);
  ```
but it now compiles correctly, but still wont connect to ttn.
I've tried entering the nwk and app keys in reverse too, still no luck!

matthias-bs, 2024-08-13:
Did you read this? [README.md#required-configuration](https://github.com/matthias-bs/BresserWeatherSensorLW/blob/main/README.md#required-configuration)
The preferred way is to add your credentials in secrets.h (just follow the example for formatting) or - if you have multiple nodes - see [README.md#loading-lorawan-node-configuration-from-file](https://github.com/matthias-bs/BresserWeatherSensorLW/blob/main/README.md#loading-lorawan-node-configuration-from-file) (again with a template which shows the formatting).

If you are still not able to join, you might have a look at the RadioLib documentation / wiki / discussion pages. I think this is a common problem. BresserWeatherSensorLW is identical to the RadioLib examples, so you might also try those first.

Do you get any hints on the TTN console?

mczakk, 2024-08-14:
i've tried a completeley fresh install of BresserWeatherSensorLW-0.9.1 and i get the error 
```/home/matt/Arduino/libraries/BresserWeatherSensorLW-0.9.1/BresserWeatherSensorLW/secrets.h:3:42: error: expected unqualified-id before numeric constant
    3 | #define RADIOLIB_LORAWAN_JOIN_EUI  0xB7, 0xC8, 0x56, 0x80, 0x23, 0x74, 0xA1, 0x69
```
line 391 0f BresserWeatherSensorLW.ino shows that the variable 'joinEUI' is a single value, not the join eui listed in 'secrets.h' (screenshot shows this)
[image]
and will not compile at all.
adding the changes to config.h and BresserWeatherSensorLW that i mentioned in my comment yesterday does make it compile ok, if you also comment out line 561 in BresserWeatherSensorLW (I'm not using LittleFS antway), just trying to determine what is being sent to TTN to join the network

mczakk, 2024-08-14:
Also, its now throwing the error
```
home/matt/Arduino/libraries/BresserWeatherSensorLW-0.9.1/BresserWeatherSensorLW/src/PayloadBresser.cpp: In member function 'void PayloadBresser::encodeWeatherSensor(int, uint8_t, LoraEncoder&)':
/home/matt/Arduino/libraries/BresserWeatherSensorLW-0.9.1/BresserWeatherSensorLW/src/PayloadBresser.cpp:376:65: error: 'struct WeatherSensor::Weather' has no member named 'wind_avg_meter_sec_fp1'; did you mean 'wind_avg_meter_sec'?
  376 |                 encoder.writeUint16(weatherSensor.sensor[idx].w.wind_avg_meter_sec_fp1);
      |                                                                 ^~~~~~~~~~~~~~~~~~~~~~
```
which is a totally new on!!

matthias-bs, 2024-08-14:
The JoinEUI and the DevEUI must be specified as uint64_t values, just as in https://github.com/matthias-bs/BresserWeatherSensorLW/blob/main/secrets.h. Just a single hex value with 16 digits. Not a comma separated array of bytes.

See https://github.com/matthias-bs/BresserWeatherSensorLW/blob/4facfd28ac5cfa705bd8d7287bd82943e6bc8fd2/src/LoadSecrets.cpp#L94 for printing the JoinEUI. It's already in the code!

It seems you have commented out the define WIND_DATA_FIXEDPOINT in WeatherSensorCfg.h

matthias-bs, 2024-08-14:
> Next update:- had to amend the config.h file
>
> ```
> // Copy over the EUI's & keys in to the something that will not compile if incorrectly formatted
> uint64_t joinEUI[8] = [...]

Well, you mixed things up in the first two lines. Please revert your changes and check the original code.

mczakk, 2024-08-15:
don't know how that happened, but i now have connection to ttn :)

trying to sort out the uplink formatter now, i get values sent to ttn, but they're not being translated by the formatter correctly

closed 2024-08-18 by mczakk

matthias-bs, 2024-08-18:
Problems solved or given up?

mczakk, 2024-08-18:
Connection solved, 😁
Opened another issue for message accessibility!