matthias-bs/BresserWeatherSensorLW #106: Bump to RadioLib 7.0

StevenCellist, 2024-09-17:
Hi @matthias-bs, just opening this so you are aware of the update to RadioLib 7.0 with some breaking changes. Most notably, `uplink()` and `downlink()` are now replaced by a single `sendReceive()` which may return a positive value when downlink is received or 0 when no downlink is received. Also, some of the return code names have been updated. There are a few more changes - I suggest taking another look at the examples provided in RadioLib (as well as the README in examples/LoRaWAN - v1.0.4 is now also officially supported). 
Feel free to close this at any moment, just a heads-up.

matthias-bs, 2024-09-17:
Hi @StevenCellist, thanks a lot for your notification and the helpful hints - and of course for putting so much effort and knowledge into the LoRaWAN part of RadioLib!

matthias-bs, 2024-09-23:
Didn't manage to update to RadioLib v7.0.0 yet. Currently, I get strange errors like `[config.h:403] debug(): Error in sendReceive - See TypeDef.h (-25)`.
Eagerly awaiting the update of [radiolib-org/radiolib-persistence](https://github.com/radiolib-org/radiolib-persistence)...

StevenCellist, 2024-09-23:
Do you have more information on this problem? (E.g. using debug protocol flag.) There has been no change to the persistence side, except for some rearranging of the internal buffer size & structure.

matthias-bs, 2024-09-24:
@StevenCellist Thanks for offering help and the information that no changes have been made regarding persistence handling.

I enabled protocol output, erased the Flash (again) to clear the Preferences, flashed the sketch, flashed my secrets.json (LoRaWAN credentials) - and it worked! (17 uplink frames without any errors)

I have no idea what was wrong before, but I suspect that the Preferences had not been cleared properly.

matthias-bs, 2024-09-24:
@StevenCellist No, unfortunately the problem has not been solved.

Logfile:
[devttyUSB0_2024_09_24.07.53.34.280.txt](https://github.com/user-attachments/files/17108879/devttyUSB0_2024_09_24.07.53.34.280.txt)

Source code:
[matthias-bs/BresserWeatherSensorLW/tree/20240922-bump-radiolib7.0](https://github.com/matthias-bs/BresserWeatherSensorLW/tree/20240922-bump-radiolib7.0)

StevenCellist, 2024-09-24:
From the log:

> RLB_PRO: ADR: no channels available for datarate 2

Is your radio asleep maybe? The channels are definitely available, so that is quite weird.

StevenCellist, 2024-09-24:
Hmmm I have a suspicion. I hope to have a look soon.

StevenCellist, 2024-09-24:
As I don't have the time to scan your full project: do you have ADR enabled? I expect it is disabled, right?

matthias-bs, 2024-09-24:
I don't call `node.setADR()` in the sketch, so ADR should be enabled (according to the example), right?

matthias-bs, 2024-09-24:
> From the log:
>> RLB_PRO: ADR: no channels available for datarate 2

> Is your radio asleep maybe? The channels are definitely available, so that is quite weird.

I don't think so - I don't put it into sleep mode between `radio.begin()` and putting the ESP32 into sleep mode.

StevenCellist, 2024-09-24:
The problem at hand is that the channel mask is either not stored correctly before going into deepsleep, or not being restored corrected after coming back from deepsleep. As you can see, the command is full of `00`:

> RLB_PRO: 00000000: 20 00 00 00 00 00 00 00 00 00 00 00 00 00

It should be either
```RLB_PRO: 00000000: 20```
or 
```RLB_PRO: 00000000: 20 ff 00 00 00 00 00 00 00 00 00 00 00 01```

With the conference tomorrow I don't have time to investigate now; I have verified that session restore works just before the release so this is a bit weird. But may be a bug after all.

matthias-bs, 2024-09-24:
Would it be helpful if I put some debug printing of the channel mask into `getBufferSession()` / `setBufferSession()`?

StevenCellist, 2024-09-24:
Yes, as well as the memcpy into the `bufferSession` in `execMacCommand` for `MAC_LINK_ADR`.

matthias-bs, 2024-09-24:
Could this (the template type in `ntoh()` / `hton()`) be a problem?

In `setBufferSession()`:

```
uint16_t chMask = LoRaWANNode::ntoh<uint32_t>(&this->bufferSession[RADIOLIB_LORAWAN_SESSION_AVAILABLE_CHANNELS]);
```

In `getBufferSession()`:

```
uint16_t chMask = 0x0000;
(void)this->getAvailableChannels(&chMask);
  LoRaWANNode::hton<uint16_t>(&this->bufferSession[RADIOLIB_LORAWAN_SESSION_AVAILABLE_CHANNELS], chMask);
```

StevenCellist, 2024-09-24:
That's an interesting typo, but this `chMask` is not the thing we're looking for here - we're looking for a 12-byte (uint64 + uint32) complete channel mask. I see the naming scheme is quite confusing.
(This channel mask tells which channels have recently been used so it doesn't reuse the same over and over randomly.)

matthias-bs, 2024-09-24:
I don't understand this entirely, but I'd say that this data is not stored in `getBufferSession()`.
In `setBufferSession()`, there seems to be the code starting at https://github.com/jgromes/RadioLib/blob/ec6de3795b1269eb051f60ae2174031048366b89/src/protocols/LoRaWAN/LoRaWAN.cpp#L517 for restoring.

StevenCellist, 2024-09-24:
MAC commands are stored in-progress in `execMacCommands`, line 1964 for ADR.

matthias-bs, 2024-09-25:
> MAC commands are stored in-progress in `execMacCommands`, line 1964 for ADR.

I hope this is correct:
```
      RADIOLIB_DEBUG_PROTOCOL_PRINTLN("chMask0123, chMask45:");
      RADIOLIB_DEBUG_PROTOCOL_HEXDUMP(&optIn[1], 12);
      memcpy(&this->bufferSession[RADIOLIB_LORAWAN_SESSION_LINK_ADR], optIn, lenIn);
```

matthias-bs, 2024-09-25:
Here is a new logfile:
[devttyUSB0_2024_09_25.07.48.32.193.txt](https://github.com/user-attachments/files/17126313/devttyUSB0_2024_09_25.07.48.32.193.txt)

StevenCellist, 2024-09-25:
In your latest logfile, all uplinks succeed and there is no error at all. So that is interesting...?

matthias-bs, 2024-09-25:
`08:36:01:010 -> RLB_PRO: PHY:  Frequency UL =   0.000 MHz` seemed suspicious to me.
I'll try to provide a better sample.

matthias-bs, 2024-09-25:
Here we go:
```
21:42:42:876 -> [ 18142][D][BresserWeatherSensorLW.ino:394] lwActivate(): Recalling LoRaWAN nonces & session
21:42:42:886 -> RLB_PRO: LoRaWAN session: v1.1
21:42:42:886 -> RLB_PRO: Setting up dynamic channels
21:42:42:891 -> RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
21:42:42:897 -> RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
21:42:42:902 -> RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
21:42:42:908 -> RLB_PRO: [MAC] 0x07
21:42:42:908 -> RLB_PRO: 00000000: 03 18 4f 84 50                                   ..O.P             
21:42:42:916 -> RLB_PRO: UL:   3 1 867.100 (0 - 5) | DL:   3 1 867.100 (0 - 5)
21:42:42:922 -> RLB_PRO: [MAC] 0x07
21:42:42:922 -> RLB_PRO: 00000000: 04 e8 56 84 50                                   ..V.P             
21:42:42:930 -> RLB_PRO: UL:   4 1 867.300 (0 - 5) | DL:   4 1 867.300 (0 - 5)
21:42:42:935 -> RLB_PRO: [MAC] 0x07
21:42:42:939 -> RLB_PRO: 00000000: 05 b8 5e 84 50                                   ..^.P             
21:42:42:946 -> RLB_PRO: UL:   5 1 867.500 (0 - 5) | DL:   5 1 867.500 (0 - 5)
21:42:42:949 -> RLB_PRO: [MAC] 0x07
21:42:42:952 -> RLB_PRO: 00000000: 06 88 66 84 50                                   ..f.P             
21:42:42:960 -> RLB_PRO: UL:   6 1 867.700 (0 - 5) | DL:   6 1 867.700 (0 - 5)
21:42:42:966 -> RLB_PRO: [MAC] 0x07
21:42:42:966 -> RLB_PRO: 00000000: 07 58 6e 84 50                                   .Xn.P             
21:42:42:974 -> RLB_PRO: UL:   7 1 867.900 (0 - 5) | DL:   7 1 867.900 (0 - 5)
21:42:42:980 -> RLB_PRO: [MAC] 0x03
21:42:42:983 -> RLB_PRO: 00000000: 20 00 00 00 00 00 00 00 00 00 00 00 00 00         .............    
21:42:42:988 -> RLB_PRO: LinkAdrReq: dataRate = 2, txSteps = 0, nbTrans = 0
21:42:42:994 -> RLB_PRO: execMacCommand(): chMask0123=0x0000000000000000
21:42:42:999 -> RLB_PRO: execMacCommand(): chMask45=0x0000
21:42:43:002 -> RLB_PRO: ADR: no channels available for datarate 2
21:42:43:009 -> RLB_PRO: LinkAdrAns: 05
21:42:43:018 -> RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
21:42:43:026 -> RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
21:42:43:031 -> RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
21:42:43:035 -> RLB_PRO: UL:   3 1 867.100 (0 - 5) | DL:   3 1 867.100 (0 - 5)
21:42:43:037 -> RLB_PRO: UL:   4 1 867.300 (0 - 5) | DL:   4 1 867.300 (0 - 5)
21:42:43:038 -> RLB_PRO: UL:   5 1 867.500 (0 - 5) | DL:   5 1 867.500 (0 - 5)
21:42:43:043 -> RLB_PRO: UL:   6 1 867.700 (0 - 5) | DL:   6 1 867.700 (0 - 5)
21:42:43:049 -> RLB_PRO: UL:   7 1 867.900 (0 - 5) | DL:   7 1 867.900 (0 - 5)
21:42:43:054 -> RLB_PRO: [MAC] 0x04
21:42:43:054 -> RLB_PRO: 00000000: 07                                               .                 
21:42:43:062 -> RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
21:42:43:065 -> RLB_PRO: [MAC] 0x05
21:42:43:068 -> RLB_PRO: 00000000: 03 d2 ad 84                                      ....              
21:42:43:076 -> RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 3, freq = 869.525
21:42:43:082 -> RLB_PRO: [MAC] 0x08
21:42:43:085 -> RLB_PRO: 00000000: 05                                               .                 
21:42:43:090 -> RLB_PRO: RXTimingSetupReq: delay = 5 sec
21:42:43:095 -> RLB_PRO: [MAC] 0x09
21:42:43:095 -> RLB_PRO: 00000000: 00                                               .                 
21:42:43:104 -> RLB_PRO: [MAC] 0x0c
21:42:43:107 -> RLB_PRO: 00000000: 65                                               e                 
21:42:43:112 -> RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
21:42:43:121 -> RLB_PRO: [MAC] 0x0f
21:42:43:122 -> RLB_PRO: 00000000: fa                                               .                 
21:42:43:126 -> RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
21:42:43:131 -> RLB_PRO: getBufferSession(): chMask=0x0073
21:42:43:137 -> [ 18394][D][BresserWeatherSensorLW.ino:416] lwActivate(): Succesfully restored session - now activating
21:42:43:154 -> [ 18413][D][BresserWeatherSensorLW.ino:627] setup(): Battery level: 0
21:42:43:154 -> [ 18419][I][BresserWeatherSensorLW.ino:633] setup(): RTC sync required
21:42:43:164 -> RLB_PRO: 
21:42:43:164 -> RLB_PRO: PHY:  Frequency UL =   0.000 MHz
21:42:43:164 -> [ 18427][D][BresserWeatherSensorLW.ino:646] setup(): Max payload length: 254
21:42:43:169 -> [ 18438][D][BresserWeatherSensorLW.ino:673] setup(): FcntUp: 2
21:42:43:174 -> [ 18444][I][BresserWeatherSensorLW.ino:722] setup(): Sending uplink; port 1, size 47
21:42:43:183 -> RLB_PRO: Uplink MAC payload:
21:42:43:186 -> RLB_PRO: 00000000: 0d                                               .                 
21:42:43:231 -> RLB_PRO: Uplink (FCntUp = 3) decoded:
21:42:43:233 -> RLB_PRO: 00000000: 5b 20 31 38 34 34 34 5d 5b 49 5d 5b 42 72 65 73  [ 18444][I][Bres
21:42:43:238 -> RLB_PRO: 00000010: 40 44 ae 0b 26 81 03 00 c3 01 3a 4c dd b5 f3 c4  @D..&.....:L....
21:42:43:247 -> RLB_PRO: 00000020: 3c dc 47 5a dc d5 e5 4c 4b 9e 1a e3 05 0b c5 2f  <.GZ...LK....../
21:42:43:255 -> RLB_PRO: 00000030: a5 28 84 d5 e4 fb e6 4a e3 a2 dc 83 74 90 2b 49  .(.....J....t.+I
21:42:43:261 -> RLB_PRO: 00000040: b7 9e eb 76 d3 ab 3a f5 98 20 31 2c 20           ...v..:.. 1,      
21:42:43:269 -> RLB_PRO: 
21:42:43:269 -> RLB_PRO: PHY:  Frequency UL =   0.000 MHz
21:42:43:274 -> [ 18534][W][config.h:403] debug(): Error in sendReceive - ERR_INVALID_FREQUENCY (-12)
21:42:43:280 -> RLB_PRO: getBufferSession(): chMask=0x0000
```

StevenCellist, 2024-09-25:
I have a strong suspicion but missing the log from the join leading up to the failure. If you can catch a complete cycle that would be perfect - the logging is awesome now - and that would likely confirm my suspicion. I don't have time to test myself yet, that would be around the weekend / early next week probably, in case you aren't in a position to continue the debugging.

matthias-bs, 2024-09-26:
I have prepended the preceding cycle to my previous post. I don't think I can do much to debug this myself.

matthias-bs, 2024-09-26:
Enjoy the conference!

StevenCellist, 2024-09-28:
@matthias-bs thanks, and fixed in https://github.com/jgromes/RadioLib/commit/5d076f616a660c371a77fa6fe40ea2f079231ec5

matthias-bs, 2024-09-28:
@StevenCellist Thanks a lot! Seems to be working now!

matthias-bs, 2024-09-28:
See https://github.com/jgromes/RadioLib/issues/1238

closed 2024-09-30 by matthias-bs