matthias-bs/BresserWeatherSensorLW #119: Tried to use the updated payload bresser.cpp to correct no rain stats, now doesnt complile

mczakk, 2024-12-12:
Hi Matthias, I saw that you had updated the payload_bresser.cpp to correct therain stats issue, so i re installed the bresserweathersensor library. now my old .ino code does not compiile at all, looking at the code, have you changed the weathersensor.h file, as the structures seem different and my code fails on those structures!

matthias-bs, 2024-12-14:
Hi @mczakk,

the first issue I see is
```
In file included from c:\Users\mtc20tfq\OneDrive - Bangor University\Documents\Arduino\libraries\RadioLib\src/modules/CC1101/CC1101.h:7,
                 from c:\Users\mtc20tfq\OneDrive - Bangor University\Documents\Arduino\libraries\RadioLib\src/RadioLib.h:69,
                 from c:\Users\mtc20tfq\OneDrive - Bangor University\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensor.h:99,
                 from C:\Users\mtc20tfq\OneDrive - Bangor University\Documents\Arduino\sketches\newbresseresp32\newbresseresp32.ino:193:
c:\users\mtc20tfq\onedrive - bangor university\documents\arduino\libraries\radiolib\src\protocols\physicallayer\physicallayer.h:138:9: error: 'FSK' conflicts with a previous declaration
  138 |   FSK = 0,
      |         ^
In file included from c:\Users\mtc20tfq\OneDrive - Bangor University\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src/lmic/lmic.h:37,
                 from c:\Users\mtc20tfq\OneDrive - Bangor University\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src/arduino_lmic.h:26,
                 from C:\Users\mtc20tfq\OneDrive - Bangor University\Documents\Arduino\sketches\newbresseresp32\newbresseresp32.ino:145:
c:\Users\mtc20tfq\OneDrive - Bangor University\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src/lmic/lorabase.h:41:14: note: previous declaration '_sf_t FSK'
   41 | enum _sf_t { FSK=0, SF7, SF8, SF9, SF10, SF11, SF12, SFrfu };
      |              ^~~
```
which shows that you are using RadioLib and MCCI_LoRaWAN_LMIC_library in the same sketch! I don't know why this happens, but this won't work!
BresserWeatherSensorLW and BresserWeatherSensorReceiver are based on RadioLib entirely. Maybe you've got some old stuff which was based on BresserWeatherSensorTTN.

For the other topic:
1. Identify the version of BresserWeatherSensorLW you created the previous fork from 
    (If the fork is still 20240821-feat-oled, you look at the file history and/or release notes around that date. Maybe it is https://github.com/matthias-bs/BresserWeatherSensorLW/releases/tag/v0.9.2 (from August 4) or https://github.com/matthias-bs/BresserWeatherSensorLW/releases/tag/v0.9.3 (from August 21)).
   
2.  Compare the changes between the version your fork was created from and the current
     (Lets assume it was v0.9.3 - then you check https://github.com/matthias-bs/BresserWeatherSensorLW/compare/v0.9.3...v0.10.1 Here you don't see any differences in using the weatherSensor structure.)

It looks as if you are using some rather old stuff, or a weird mixture of different versions. The way you are using the structure dates back prior to changes made in https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/caac416004e4d83992c334aad89fcb8635505f5a (27 Oct 2023).

Looking at [package.json](https://github.com/matthias-bs/BresserWeatherSensorLW/blob/main/package.json), you always see a consistent set of library versions which is known to work.

mczakk, 2024-12-21:
yes, i probably have old stuff that was based on BresserWeatherSensorTTN as i havent updated it since the discovery that the rain statistyics were not working properly.
I have downlaoded you latest version  (LW) and everything seems very different. I cant see how to integrate the SSD1306 display in the new .ino  and when i try and compile without it i am getting all sorts of errors relating to BLE sensors and 'nimbleutils'
I've created a repository called old-code, which i think is the last working version of the esp bresser with oled (but failed tto do the rain stats). can you incorporate the display elements of this into the current (LW) code?

Many Thanks
Matt